PodoPrinter/Cura #11: Adding hte gantry to Cura?

riktheunissen, 2018-08-23:
I added a platform to Cura representing the BlackBelt with gantry (screenshot). I think it makes more sense to the slicing preview. 

What do the rest of you think of this?

[image]

**Application Version**
3.4.3 and up

**Platform**
all

**Printer**
BlackBelt 3D

Thanks,
Rik Theunissen
Application Development @ BlackBelt 3D

BBStephan, 2018-08-23:
@fieldOfView - Can we use different Mockups for different Gantry Angles?

fieldOfView, 2018-08-23:
I will have to check to see if the model can be easily (ex)changed when changing the gantry angle.

BBStephan, 2018-09-11:
Looks awesome, indeed! But to me it only makes sense if we can display the correct printing angle (15°/25°/35°/45°).