PodoPrinter/Cura #12: Layer view, Feedrate visual misplacement

riktheunissen, 2018-09-05:
**Application Version**
3.4.3 test version

**Platform**
windows

**Printer**
BlackBelt 3D printer

**Steps to Reproduce**
Different stl files reproduce the same result. 
Reinstalled Cura.
Opend hte Gcode in Simplify3D and the speed was fine.

**Actual Results**
It shows a different speed to the support lines than is in the G-Code. 

**Expected results**
Only the belt wall should be dark blue since this is the only 10 mm/s line.

**Additional Information**
[image]

fieldOfView, 2018-09-10:
Can you attach a project file (or send it over Slack)?

riktheunissen, 2018-09-11:
It's on the Slack channel