PodoPrinter/Cura #15: No firmware update possible

riktheunissen, 2018-09-24:
**Application Version**
3.5 alpha

**Platform**
windows

**Printer**
BlackBelt 3D printer

**Steps to Reproduce**
Connecting the printer to the laptop in order to update the firmware. 
- Printer power on
- USB in printer and laptop
- Printer board reboots
- Windows recognizes connection
- Opening Cura and go to Printer menu.
- Select firmware upgrade 

**Actual Results**
no connection possible

**Expected results**
An active connection to auto update the firmware.

**Additional Information**

fieldOfView, 2018-09-24:
Could you try waiting a little longer between starting Cura and trying to upload firmware? Cura needs to find out the baudrate to use, and this takes up to half a minute. Once there is a connection, a blue icon should appear next to the printer name on top of the sidebar.

riktheunissen, 2018-09-24:
Tryed it and even after 5 minutes no connection.

fieldOfView, 2018-09-24:
I can reproduce the issue, with the build I sent you, but not when I run it in my development environment. I'll see if I can find out what (doesn't) happen anyway.

fieldOfView, 2018-09-24:
Looks like I already fixed this for Ultimaker Cura, but somehow the fix got omitted from the BlackBelt build:
https://github.com/Ultimaker/Cura/pull/4370

riktheunissen, 2018-09-24:
Seems to work now, which version of the firmware is uploaded?

fieldOfView, 2018-09-24:
1.1.8, same version that was uploaded in 3.4