PodoPrinter/Cura #17: Crash on custom support change

riktheunissen, 2018-09-25:
**Application Version**
3.5 beta

**Platform**
W10

**Printer**
BB3D

**Steps to Reproduce**
open it again, Old projects still work 
projects crash on change
Even moving it slightly doesn't affect it

**Actual Results**
Cura shuts down and corrupts the config

**Expected results**
change to support structure

**Additional Information**
[35_04_PLA_Profile_100.zip](https://github.com/BlackBelt3D/Cura/files/2414818/35_04_PLA_Profile_100.zip)

fieldOfView, 2018-09-25:
Could you please clarify? Your first step to reproduce is "open it again". Open what again? I don't understand what you are trying to say.

riktheunissen, 2018-09-25:
updated the steps

riktheunissen, 2018-09-25:
actually it just closes after every change of mesh type to "print as support". Maybe it is not a crash. I added a log of a random file changed mesh type to "print as support".

fieldOfView, 2018-09-25:
Thanks, much clearer now. I have reproduced the same issue in Ultimaker Cura 3.5 (with any printer), so I will report it in the main project while trying to find a fix.

fieldOfView, 2018-09-28:
Has been fixed for 3.5

closed 2018-09-28 by fieldOfView