PodoPrinter/Cura #27: Preferred changes to several settings

riktheunissen, 2018-12-12:
**Application Version**
3.6.0

**Platform**
W10

**Printer**
BB 3D

**Steps to Reproduce**
Made a fresh install and noticed the following settings:

**Actual Results**

1. the outer line width is always a 0.1 wider than the nozzle.
2. Remove Mesh intersection is always off
3. Naming of materials results in "copolyester" now
4. Raft flow is always on 100%
5. Raft air gap = 0.2 mm
6. Fill gaps between walls: everywhere
7. Belt wall offset can be set below 0.25 and nothing happens

**Expected results**
1. the outer line width = the nozzle width
2. Remove Mesh intersection is always on
3. Naming of materials results in "PETG, nGen, HT" now
4. Raft flow = Material flow
5. Raft air gap = 0.5 mm
6. Fill gaps between walls: nowhere
7. The field should turn red indicating that below 0.25 is not recommended and can be harmful to the belt.

**Additional Information**

fieldOfView, 2018-12-12:
These changes should be the same for all materials?

BBStephan, 2018-12-12:
Yes please :)

fieldOfView, 2018-12-12:
Could you clarify what you mean by "Naming of materials results in "copolyester" now"? Where/how is it confusing? (it is not wrong, right?) Setting the Material Type to "Copolyester" was requested in the "BB-Cura 3.6 slicer settings" spreadsheet I was given on slack on november 29th. Should I revert that?

BBStephan, 2018-12-12:
Sorry, we've not been clear about that.
The material name in the gcode name extension needs to be: nGen, HT, PETG, PLA and TPU.
The last two are good, but the others are being labeled as "copolyester".

fieldOfView, 2018-12-12:
Ah, I had not considered the job name prefix (which is used for the gcode filename).

closed 2019-02-05 by fieldOfView