PodoPrinter/Cura #30: Filament Diameter

riktheunissen, 2019-03-21:
**Application Version**
BB Cura 3.6.0

**Platform**
W10

**Printer**
BlackBelt 3D DD

**Steps to Reproduce**
Printed several times and changed the diameter in Cura to 2.85 as well as 1.75. Works best wit a 1.75 but should work with 2.85 mm settings.

**Actual Results**
Under extrusion or overflow. Setting flow to 300% on the machine seems to work. With the 1.75 we get over extrusion.

**Expected results**
To print as good as it does with the 1.75 installed. I think the filament diameter is not taken into consideration in the corrections.

**Additional Information**

[35_04_Cheetah_sample02_support.zip](https://github.com/BlackBelt3D/Cura/files/2992499/35_04_Cheetah_sample02_support.zip)

BBStephan, 2019-03-21:
Is there a downside of adjusting the flow multiplier?

riktheunissen, 2019-04-01:
Not sure, but I guess the retraction and other compensation based on length are then also reduce or increased.

closed 2019-04-01 by BBStephan