PodoPrinter/Cura #32: Command line usage for integration in to workflow.

mmiscool, 2019-08-20:
**Application Version**
Any

**Platform**
Linux Headless environment 

**Printer**
Blackbelt and variants. 

Hello,
I am with Autodrop3d. We have been working on cloud based management software for our own auto ejecting printers. 

We were hoping to support Blackbelt style machines in the near future as endpoint devices. 

Would be interested in getting some help interfacing with the slicer, Pre-processor and post processor. The blackbelt software is a unique configuration. If it can be made to work from the command line so that it can be run in cloud environment that would be amazing.

fieldOfView, 2020-09-14:
Blackbelt Cura uses the "normal" Curaengine, but there is some preprocessing going on in the front-end to prepare/transform geometry before it is sliced. Some features also require postprocessing the generated gcode. Extracting the preprocessing and postprocessing steps is not impossible, but is not a trivial undertaking.

closed 2020-09-14 by fieldOfView

Wiliamoliveira1, 2020-09-14:
Will you create a new slicer or an update similar to BBCURA 3.6.0 (U1)?
I realized that you will use vanilla
We can help with more people programming and a little $
Remember that it is open source

mmiscool, 2020-09-28:
I was able to get some funding together to support an effort to make a command line version.