PodoPrinter/Cura #33: A problem occurred at runtime, When I trying to run  BlackBelt Cura from source code.

ShenRen, 2019-12-23:
**Brief Description**
    Recently I have been studying tilting printing .BlackBelt Cura is a very useful program. So. I want run whole BlackBelt Cura from source code . But now I'm at my wit's end.
    It is my understanding that Uranium module is incompatible. But I  don't know where to find the right version.

**Application Version**

branch  bb-3.6

**Platform**

Windows 10  pro 64bit

**Printer**

BlackBelt

**Steps to Reproduce**

First , I have succeeded run  Cura3.6 from source code. Everything is work fine.
Second ,original cura 3.6 code was replaceed  by branch  bb-3.6 ,original  fdm_materials was replaceed  by   BlackBelt3D`s  fdm_materials  too.  Application can run , but can not slice.

**Actual Results**

[cura.log](https://github.com/BlackBelt3D/Cura/files/3995484/cura.log)
An error occurred while running. Log shows here:

DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [169]: Begin loading of script: PauseAtHeightforRepetier
DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [169]: Begin loading of script: SearchAndReplace
DEBUG - [MainThread] PostProcessingPlugin.PostProcessingPlugin.loadScripts [169]: Begin loading of script: Stretch
DEBUG - [MainThread] UM.Logger.logException [84]: Exception: An exception occurred in inherit function =extruderValue(adhesion_extruder_nr, 'adhesion_type'): '0'
DEBUG - [MainThread] UM.Logger.logException [88]: Traceback (most recent call last):
DEBUG - [MainThread] UM.Logger.logException [88]:   File "F:\Cura360_TiltPrint_20191222\Download\Uranium\UM\Settings\SettingFunction.py", line 95, in __call__
DEBUG - [MainThread] UM.Logger.logException [88]:     return eval(self._compiled, g, locals)
DEBUG - [MainThread] UM.Logger.logException [88]:   File "<UM.Settings.SettingFunction (0x23e624d8e10) =extruderValue(adhesion_extruder_nr, 'adhesion_type') >", line 1, in <module>
DEBUG - [MainThread] UM.Logger.logException [88]:   File "F:\Cura360_TiltPrint_20191222\Download\Cura\cura\Settings\CuraFormulaFunctions.py", line 41, in getValueInExtruder
DEBUG - [MainThread] UM.Logger.logException [88]:     extruder_stack = global_stack.extruders[str(extruder_position)]
DEBUG - [MainThread] UM.Logger.logException [88]: KeyError: '0'
DEBUG - [MainThread] UM.Logger.logException [84]: Exception: An exception occurred in inherit function =extruderValue(adhesion_extruder_nr, 'adhesion_type'): '0'
DEBUG - [MainThread] UM.Logger.logException [88]: Traceback (most recent call last):
DEBUG - [MainThread] UM.Logger.logException [88]:   File "F:\Cura360_TiltPrint_20191222\Download\Uranium\UM\Settings\SettingFunction.py", line 95, in __call__
DEBUG - [MainThread] UM.Logger.logException [88]:     return eval(self._compiled, g, locals)
DEBUG - [MainThread] UM.Logger.logException [88]:   File "<UM.Settings.SettingFunction (0x23e624d8e10) =extruderValue(adhesion_extruder_nr, 'adhesion_type') >", line 1, in <module>
DEBUG - [MainThread] UM.Logger.logException [88]:   File "F:\Cura360_TiltPrint_20191222\Download\Cura\cura\Settings\CuraFormulaFunctions.py", line 41, in getValueInExtruder
DEBUG - [MainThread] UM.Logger.logException [88]:     extruder_stack = global_stack.extruders[str(extruder_position)]
DEBUG - [MainThread] UM.Logger.logException [88]: KeyError: '0'
DEBUG - [Thread-1] UM.Mesh.MeshData.calculateNormalsFromVertices [444]: Calculating normals took 0.0019943714141845703 seconds
DEBUG - [Thread-1] STLReader.STLReader._read [89]: Loaded a mesh with 82872 vertices
DEBUG - [MainThread] UM.Logger.logException [84]: Exception: An exception occurred in inherit function =extruderValue(adhesion_extruder_nr, 'adhesion_type'): '0'
DEBUG - [MainThread] UM.Logger.logException [88]: Traceback (most recent call last):
DEBUG - [MainThread] UM.Logger.logException [88]:   File "F:\Cura360_TiltPrint_20191222\Download\Uranium\UM\Settings\SettingFunction.py", line 95, in __call__
DEBUG - [MainThread] UM.Logger.logException [88]:     return eval(self._compiled, g, locals)
DEBUG - [MainThread] UM.Logger.logException [88]:   File "<UM.Settings.SettingFunction (0x23e624d8e10) =extruderValue(adhesion_extruder_nr, 'adhesion_type') >", line 1, in <module>
DEBUG - [MainThread] UM.Logger.logException [88]:   File "F:\Cura360_TiltPrint_20191222\Download\Cura\cura\Settings\CuraFormulaFunctions.py", line 41, in getValueInExtruder
DEBUG - [MainThread] UM.Logger.logException [88]:     extruder_stack = global_stack.extruders[str(extruder_position)]
DEBUG - [MainThread] UM.Logger.logException [88]: KeyError: '0'
INFO - [MainThread] cura.Settings.GlobalStack.addExtruder [90]: Extruder[blackbelt_extruder_0 #2] added to [BLACKBELT 3D Printer] at position [0]
WARNING - [MainThread] cura.Settings.ExtruderManager.registerExtruder [168]: Extruder [blackbelt_extruder_0 #2] has already been registered on machine [BLACKBELT 3D Printer], not doing anything
WARNING - [MainThread] cura.Machines.MaterialManager.getFallbackMaterialIdByMaterialType [488]: The material type [Copolyester] does not have a fallback material
INFO - [MainThread] cura.Settings.MachineManager._initMachineState [342]: Machine 'BLACKBELT 3D Printer' quality set to 'normal'

When slice , The background commands are not quite right either. No settings transfer to slice engine.

INFO - [MainThread] UM.Backend.Backend.startEngine [74]: Started engine process: F:\Cura360_TiltPrint_20191222\Cura3.6\CuraEngine.exe
DEBUG - [MainThread] UM.Backend.Backend._backendLog [94]: [Backend] Calling engine with: ['F:\\Cura360_TiltPrint_20191222\\Cura3.6\\CuraEngine.exe', 'connect', '127.0.0.1:49674', '']
DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onBackendQuit [809]: Backend quit with return code 3221225781. Resetting process and socket.
DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onStartSliceCompleted [430]: Sending slice message took 0.3480682373046875 seconds

**Expected results**

**Additional Information**

IDE : PyCharm
Python : 3.5

Some screenshots:
[image]
[image]
[image]

JaideepBgit, 2020-09-14:
Hi @ShenRen , I have tried to run BlackBelt Cura from source code on Ubuntu platform, but it crashed for me with 2 errors:
------------------------------------------------------------------------------------------------------------------------------
Few plugins are not loaded when run cura_app.py
-----------------------------------------------------------------
[MainThread] UM.PluginRegistry.loadPlugin [417]: Plugin [RotateTool] with supported sdk versions [['7.3.0']] is incompatible with the current sdk version [5.0.0].
[MainThread] UM.PluginRegistry.loadPlugin [417]: Plugin [CameraTool] with supported sdk versions [['7.3.0']] is incompatible with the current sdk version [5.0.0].
[MainThread] UM.PluginRegistry.loadPlugin [417]: Plugin [ScaleTool] with supported sdk versions [['7.3.0']] is incompatible with the current sdk version [5.0.0].
[MainThread] UM.PluginRegistry.loadPlugin [417]: Plugin [SelectionTool] with supported sdk versions [['7.3.0']] is incompatible with the current sdk version [5.0.0].
[MainThread] UM.Backend.Backend._logSocketState [192]: Socket state changed to Connected
[MainThread] UM.Backend.Backend._onSocketStateChanged [181]: Backend connected on port 49679
[MainThread] UM.PluginRegistry.loadPlugin [417]: Plugin [MirrorTool] with supported sdk versions [['7.3.0']] is incompatible with the current sdk version [5.0.0].
[MainThread] UM.PluginRegistry.loadPlugin [417]: Plugin [TranslateTool] with supported sdk versions [['7.3.0']] is incompatible with the current sdk version [5.0.0].
[MainThread] UM.PluginRegistry.loadPlugin [405]: Plugin [UpdateChecker] has been disabled. Skip loading it.
[MainThread] UM.PluginRegistry.loadPlugin [417]: Plugin [LocalFileOutputDevice] with supported sdk versions [['7.3.0']] is incompatible with the current sdk version [5.0.0].
---------------------------------------------------------------------------------------------------------------------------------------------
I have bypassed the plugins version incompatibility and loaded all plugins. Later it was loading the UI, but it crashed with no errors but with few errors or warnings on the console
[cura.log](https://github.com/BlackBelt3D/Cura/files/5219272/cura.log)

Cyclic dependency detected between "file:///home/ifactory3d/Documents/work/bb-Cura-bb-3.6/resources/qml/Actions.qml" and "file:///home/ifactory3d/Documents/work/bb-Cura-bb-3.6/resources/qml/Actions.qml"
QQmlApplicationEngine failed to load component
2020-09-14 07:37:45,707 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///home/ifactory3d/Documents/work/bb-Cura-bb-3.6/resources/qml/Cura.qml:938:5: Type AddMachineDialog unavailable
2020-09-14 07:37:45,708 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [406]: file:///home/ifactory3d/Documents/work/bb-Cura-bb-3.6/resources/qml/AddMachineDialog.qml:76:17: Cannot assign to non-existent property "preferredSectionValue"

fieldOfView, 2020-09-14:
You need to use the 3.6 branch of Uranium.

fieldOfView, 2020-09-14:
There is no support for running this version of Cura from source; only the official releases are supported.

closed 2020-09-14 by fieldOfView

JaideepBgit, 2020-09-14:
Where do I get the 3.6 branch of Uranium

fieldOfView, 2020-09-14:
See https://github.com/Ultimaker/Uranium/tree/3.6