PodoPrinter/Cura #34: [3.6.0] Blackblet Cura version 3.6 does not open the application when cura_app.py is run.

JaideepBgit, 2020-09-14:
**Application Version**
Blackbelt Cura 3.6

**Platform**
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.1 LTS
Release:	20.04
Codename:	focal

**Steps to Reproduce**
It is the cura build and run steps found on [https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source-on-Ubuntu](url).
Finally when "python3 cura_app.py" is run, it is not loading the application.

**Actual Results**
Attached in the log. It is loading the plugins but is timed out and not opening the Blackbelt Cura Application

**Expected results**
Should run the BlackBelt Cura Application

**Additional Information**
Attached is the log for your information:
[cura.log](https://github.com/BlackBelt3D/Cura/files/5219422/cura.log)

fieldOfView, 2020-09-14:
Only the AppImage provided by Blackbelt is supported on Linux.

closed 2020-09-14 by fieldOfView

JaideepBgit, 2020-09-14:
On which Operating system can I run the cura_app.py from source?

fieldOfView, 2020-09-14:
You can run it on any operating system that is capable of running Cura, but you will not get support for it here.