PodoPrinter/Cura #35: [3.6] Blackbelt CURA application is not slicing the

JaideepBgit, 2020-09-15:
**Application Version**
Blackbelt Cura 3.6

**Platform**
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.1 LTS
Release:	20.04
Codename:	focal

**Steps to Reproduce**
It is the cura build and run steps found on [https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source-on-Ubuntu](url).
Finally when "python3 cura_app.py" is run and app opens. When attempted to slice it just hangs and 

**Actual Results**
Attached in the log. It is loading the stl code, but it is not slicing the object.

**Expected results**
Should run slicing on the BlackBelt Cura Application

**Additional Information**
Attached is the log for your information:
[cura.log](https://github.com/BlackBelt3D/Cura/files/5223260/cura.log)

fieldOfView, 2020-09-15:
Thanks for your interest in BlackBelt Cura. Remember that only the AppImage provided by Blackbelt is supported on Linux.

If you want my help, please provide the full cura.log, not just the part that you think is relevant. I have a hunch that you may have built a version of CuraEngine that is too new for BlackBelt Cura. You need the 3.6 branch of CuraEngine.

closed 2020-09-15 by fieldOfView

JaideepBgit, 2020-09-15:
Hi @fieldOfView , After you mentioned as above I tried with CURAEngine 3.6 version but it is not building due to lot of errors.

JaideepBgit, 2020-09-15:
@fieldOfView. I have checked from version 3.6 to 4.3 and could only build the 4.3 as all the errors have been resolved in it. But slicing didn't work.
As requested I have attached the complete cura log for your reference.
[cura_1.log](https://github.com/BlackBelt3D/Cura/files/5223861/cura_1.log)

fieldOfView, 2020-09-15:
Blackbelt Cura 3.6 is only ever going to work with a 3.6 version of CuraEngine, which also requires a 3.6 version of libArcus and possibly other dependencies.

JaideepBgit, 2020-09-15:
Hi @fieldOfView. I have checked and installed the versions of dependencies as requested. The 3.6 version of CuraEngine doesn't get build because of the programming issues. What was the version of GCC which was used by you?  as the code is crashing when building the .cpp code. If possible can you try once building the  3.6 version of CuraEngine in your linux environment.
[curaengine_build.log](https://github.com/BlackBelt3D/Cura/files/5224075/curaengine_build.log)

fieldOfView, 2020-09-15:
For the Blackbelt Cura project, I don't build CuraEngine but I use the version of CuraEngine that comes with the Ultimaker Cura 3.6 AppImage. I repackage that version with the Blackbelt specific frontend code.

JaideepBgit, 2020-09-15:
@fieldOfView I have tried to use Ultimaker Cura 3.6 AppImage for CuraEngine. But the curaengine is of different executable format. Can you share the Blackbelt cura 3.6 app image for linux or the CuraEngine of Ultimaker Cura 3.6 app image in executable format

fieldOfView, 2020-09-15:
See https://blackbelt-3d.com/pub/software/BlackBelt_Cura_3.6.0/BLACKBELT_Cura_3.6.AppImage

Note that I am not a BlackBelt employee; I am the external contractor who made BlackBelt Cura for them.

fieldOfView, 2020-09-17:
You could try cura-build (https://github.com/Ultimaker/cura-build/), or do what I did for BlackBelt Cura which is unpackage the Ultimaker AppImage, and repackage it with the changed bits using AppImageKit (https://github.com/AppImage/AppImageKit).

JaideepBgit, 2020-09-17:
@fieldOfView . I didn't understand the meaning of changed bits. Sorry to ask many questions.

I'll rephrase my question again:
a) I made a small change in Blackbelt cura and added an extra plugin for my project.
b) Now I want to create an AppImage for it. 
c) For the suggestion provided by you, I could use your method.
    c.1) If I added a new property in the Qt application, how can I compile it a library or .so file or similar one.

I'm sorry, but I didn't actually understand the process of creating an AppImage.

Your help once again would be of great help.

fieldOfView, 2020-09-17:
I am sorry, I don't currently have time to walk you through this.