PodoPrinter/Cura #38: NMake does not build all dependencies

JaideepBgit, 2020-09-24:
**Application Version**
BlackBelt CURA 3.6 version

**Platform**
Processor: Intel(R) Core(TM) i5-8300H CPU @ 2.30 GHz
RAM: 16.0 GB
System type: 64-bit Operating system, x64-based processor

**Printer**
Blackbelt Printer
 
**Steps to Reproduce**
I am trying to run source from [https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source-on-Windows#microsoft-visual-studio-2019](url). 
And attempting to run step 6.
6. Open a terminal and run NMake in the build directory. If all goes well, this should build all dependencies.

**Actual Results**
PS D:\work\cura-build-environment-master\build> NMake

Microsoft (R) Program Maintenance Utility Version 12.00.21005.1
Copyright (C) Microsoft Corporation.  All rights reserved.

NMAKE : fatal error U1064: MAKEFILE not found and no target specified
Stop.

**Expected results**
Should build all dependencies.

fieldOfView, 2020-09-24:
I don't know how many times I can tell you this; only the official builds are supported. If you need support building Cura, refer to the Ultimaker repository instead.

closed 2020-09-24 by fieldOfView

JaideepBgit, 2020-09-25:
https://github.com/Ultimaker/Cura/issues/8435

it has the solution