PodoPrinter/Cura #40: Not able to slice object in Blackbelt CURA. It is hanging.

JaideepBgit, 2020-10-05:
**Application Version**
Blackbelt CURA 3.6 version

**Platform**
Operating system: Windows 10
Processor: Intel(R) Core(TM) i5-8300H CPU @ 2.30 GHz
RAM: 16.0 GB
System type: 64-bit Operating system, x64-based processor

 
**Steps to Reproduce**
I was able to run Blackbelt CURA from source on windows operating system.  Unable to slice and it hangs. I have installed all libraries like Arcus, Savitar, Curaengine of version 3.6. I have attached the log with the error.

**Actual Results**

2020-10-05 16:53:36,175 - DEBUG - [Thread-14] UM.Backend.Backend._backendLog [94]: [Backend] [ERROR] Trying to retrieve setting with no value given: 'machine_always_write_active_tool'
2020-10-05 16:53:36,268 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [186]: Socket state changed to Closing
2020-10-05 16:53:36,272 - DEBUG - [MainThread] UM.Backend.Backend._onSocketError [208]: Socket debug: Arcus Error (13): Closing socket because other side requested close.
2020-10-05 16:53:36,276 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [188]: Socket state changed to Closed
2020-10-05 16:53:36,280 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onBackendQuit [809]: Backend quit with return code 2. Resetting process and socket.
qml: TableViewSelection: index out of range
qml: TableViewSelection: index out of range
2020-10-05 17:01:26,938 - DEBUG - [MainThread] cura.AutoSave._onTimeout [48]: Autosaving preferences, instances and profiles

**Expected results**
Should slice the object.

fieldOfView, 2020-10-05:
I don't know how many times I can tell you this; only the official builds are supported.

closed 2020-10-05 by fieldOfView

JaideepBgit, 2020-10-05:
Oh , I understand. I want to ask one more question, in ubuntu I was able to run with the 3.6 version of the libraries, but not in windows. 
CURA didn't raise an error, but Blackbelt CURA did. Hence, I request for a small suggestion in this regard

fieldOfView, 2020-10-05:
Blackbelt Cura uses the Ultimaker-built version of CuraEngine version 3.6. Test if the binary CuraEngine.exe from the original distribution works. If it does, you are doing something wrong building CuraEngine. I have no idea what.

fieldOfView, 2020-10-05:
It seems you have built the 4.7 or master branch of CuraEngine. You need the 3.6 branch.

The 'machine_always_write_active_tool' setting CuraEngine complains about being missing in your definitions was not added before Cura 4.7.

fieldOfView, 2020-10-05:
Out of curiosity, what printer are you building a Cura-version for?

JaideepBgit, 2020-10-06:
Hi @fieldOfView, That is really good question, I don't know how to name it but I'm building a custom printer for the CURA version. 

I would like to thank you for the suggestion to use CuraEngine from the original distribution and the error 'machine_always_write_active_tool' is solved. 

But still the slicing is hanging and there is no error in the log:
QWindowsNativeFileDialogBase::onSelectionChange () 0
QWindowsNativeFileDialogBase::onSelectionChange () 0
QWindowsNativeFileDialogBase::onSelectionChange () 0
QWindowsNativeFileDialogBase::onSelectionChange (QUrl("file:///C:/dev/XYZ_20mm_Calibration_Cube/files/xyzCalibration_cube.stl")) 1
2020-10-06 14:03:43,067 - DEBUG - [Thread-2] UM.Mesh.MeshData.calculateNormalsFromVertices [444]: Calculating normals took 0.0 seconds
2020-10-06 14:03:43,068 - DEBUG - [Thread-2] STLReader.STLReader._read [89]: Loaded a mesh with 408 vertices
2020-10-06 14:03:43,069 - DEBUG - [MainThread] UM.Mesh.MeshData.approximateConvexHull [401]: approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.0009984970092773438 seconds. 8 input vertices. 8 output vertices.
2020-10-06 14:03:43,152 - DEBUG - [Thread-2] UM.FileHandler.ReadFileJob.run [65]: Loading file took 0.1 seconds
2020-10-06 14:03:43,153 - DEBUG - [Thread-2] UM.Mesh.MeshData.approximateConvexHull [401]: approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.0010025501251220703 seconds. 8 input vertices. 8 output vertices.
2020-10-06 14:03:43,876 - DEBUG - [MainThread] UM.Settings.ContainerRegistry.addContainer [371]: Container [per_object_stack_1940651381424] added.
2020-10-06 14:03:44,202 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [55]: Loading shader file [D:\work\secondtry\Cura-bb-3.6\cura\..\resources\shaders\transparent_object.shader]...
2020-10-06 14:04:05,588 - DEBUG - [MainThread] UM.View.GL.ShaderProgram.load [55]: Loading shader file [D:\work\secondtry\Uranium-3.6\UM\..\resources\shaders\toolhandle.shader]...
2020-10-06 14:04:16,106 - DEBUG - [MainThread] BlackBeltPlugin.CuraEngineBackendPatches.slice [25]: Starting to slice...
2020-10-06 14:04:16,107 - DEBUG - [MainThread] BlackBeltPlugin.CuraEngineBackendPatches.slice [42]: Going to slice build plate [0]!
2020-10-06 14:04:16,707 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onStartSliceCompleted [430]: Sending slice message took 0.6005020141601562 seconds
2020-10-06 14:07:24,741 - DEBUG - [MainThread] cura.AutoSave._onTimeout [48]: Autosaving preferences, instances and profiles

JaideepBgit, 2020-10-06:
Can you please provide any suggestion with regard to log above.